raspberrypi/pico-project-generator #5: Doesn't work in WSL because tkinter is required

nmk456, 2021-01-23:
Running pico_project.py in WSL fails because there's no GUI and therefore no tkinter module. This could be fixed perhaps by only importing tkinter if pico_project.py is run with the GUI option.

JamesH65, 2021-01-23:
Interesting. I didn't try WSL, just Linux (x86 and Arm), Windows and Mac. I know nothing of WSL, but presumably installing tkinter library is not a possibility if there is no GUI. Will have a think.

joshuag1000, 2021-01-23:
WSL does not support GUI's at all yet. Microsoft plan to add this feature in the future. If you are going to want to use a GUI then you are going to need to run it on windows itself by installing python.

JamesH65, 2021-01-23:
Had a quick investigate, and can conditionally import, so will sort that out on Monday.

joshuag1000, 2021-01-23:
After a quick google i was able to find out that you can install tkinter using: `sudo apt-get install python3-tk`.
I was able to install that on my version of windows subsystem for linux and i was able to get the script to run.
I have not done much testing so i cannot say for sure if this works 100%. 
I will report back when i have done some more testing

joshuag1000, 2021-01-23:
Ok so far I have had no issues after insalling the module. Looks like that has appeared to fix the issue @nmk456 .

JamesH65, 2021-01-23:
I'll still fix it up, shouldn't need tkinter installed if only running from the command line.

JamesH65, 2021-01-25:
Hmm, bit more of a PITA that initially thought. Without TKInter a load of top level classes break the rest of the code as tk is undefined. Will need to swot up a bit more on Python and imports etc.

joshuag1000, 2021-01-25:
am i correct in thinking tkinter is only needed for loading the gui?

joshuag1000, 2021-01-25:
If so then can you just use an if statement to import the tkinter module only when the gui option is called?

joshuag1000, 2021-01-25:
Ok this is a bit messy and definatly is not the neatest solution. but i have thrown together a "patch" based on my comment/thought before that basically only imports and runs code that needs tkinter when the --gui or -g option is called. 
I have not done massive amounts of testing but it does appear to work.
I did have to reshuffle some lines of code but nothing has been removed.
I hope this is a little help to build a neater solution. there is a chance i broke something but i have not spotted any bugs atm.
I have attached a zip of these modifications of this project with the possible patch.

SHA-256: E6F5CD91BD23A4895175F5BDEEDD540B2075C339C32B8E64CC7EC4C646C78780
[pico-project-generator-MODIFIED.zip](https://github.com/raspberrypi/pico-project-generator/files/5867849/pico-project-generator-MODIFIED.zip)
VirusTotal of the file for security: https://www.virustotal.com/gui/file-analysis/Mzk1NDk3OGQxNmEzMzIzMmZlMzA5M2Y1ZDA2MGU4NjA6MTYxMTU5MTQxOQ==/detection

JamesH65, 2021-01-25:
I was trying something along those lines. The mass movement of classes is a PITA. Will do something similar and try and maintain some semblance of code sanity!

JamesH65, 2021-01-25:
Note, also need to surround the TK import in try catch because you still need to catch people asking for a GUI when its not supported.

joshuag1000, 2021-01-25:
Ok so I have done a massive restructure of the code... It took a while. Nothing has changed only how the code is stored and how different things are called. The restructure allows for the program to work fine without tkinter without a GUI. 
The restructure also allows possible easier expansion of the code. The restructure splits the code into 4 python files: 
**pico_project.py** - This is the start file (Like normal) this is where the program enters and receives any parameters.
**pico_project_generation.py** - This file was needed when ripped the GUI from the main file to prevent import loops and incomplete import errors.
**pico_project_settings.py** - in here are all the variables that are needed by all of the other files in the project. I also put things like the code fragments and lists to make them easy to find when adding more. Again, this was needed so the main 2 files pico_project and pico_project_gui can access all the variables needed.
**pico_project_gui.py** - This contains ALL code needed for the gui. If the user wanted to they could drop this file entirely and the project would/should work fine without a GUI.
If you decide to merge this code, I would highly recommend testing my code and double checking it before merging it with the master.
The code also is not the neatest, but I have tried my best!
I really hope this helps at least a little!
I have created a pull request with the code so you can easily see it.
Once again i have done some testing on both windows (normal python install with tkinter) and WSL with the tkinter module uninstalled.
Pull request: #11

joshuag1000, 2021-01-25:
> Note, also need to surround the TK import in try catch because you still need to catch people asking for a GUI when its not supported.

Oh while i remember i forgot to add this into my code. but it shouldn't matter too much because if they are trying to run a gui in an environment that does not have one installed or supported then that will be the only time the program crashes.

JamesH65, 2021-01-26:
I'll take a look as son as I can, I suspect a change this large will need a loot of checking over, but its along the lines I  was thinking of, so useful. Not how how splitting it up will affects its running in all circumstances.

joshuag1000, 2021-01-26:
Oh yeah i agree a change that massive will needs lots of looking over! It wont necessarily affect the running in most circumstances but it will make the code easier to maintain and having the settings inside one file makes changing them a little easier.

lurch, 2021-05-08:
> Just so most of us are going to use WSL for development

I don't know where you got that idea? We (Raspberry Pi) don't provide any instructions for Pico development using WSL, and therefore we don't officially support it. But if you want to use it, and it works for you, then great :+1: :slightly_smiling_face:

berrak, 2021-07-12:
Looks that the `tkinter`module installation itself, or rather the documentation for the installation on Linux based platforms (Ubuntu/Debian/RPI) has cause some unnecessary issues over time.
I submitted a simple PR for the README which fix that issue.

mklemarczyk, 2023-05-20:
Could the library tkinter be loaded only in GUI mode? It does not bring any value in CLI mode.
Python is dynamic in the library loading, we could define that GUI parts are not loaded in CLI mode of script.