raspberrypi/pico-project-generator #13: macOS Mojave logout at end of generation process

HeadBoffin, 2021-01-28:
macOS 10.14.6, Python 3.8.5

When the GUI has finished creating a new project, something happens to cause the user to be logged out of macOS.

I found these stray lines in Terminal on the first time I logged back in:

```
Mojo:pico-project-generator nick$ python3 pico_project.py --gui
objc[72662]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff90cad3f0) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x10cb0af50). One of the two will be used. Which one is undefined.
```

Project was created fine along with the random HW divider example, but I'll get to debug this later.

Do not action in a hurry, I'll instrument my pico_project.py for this Mac and I'll try it on another one, but if I don't note it now, it will get lost for a few days.

HeadBoffin, 2021-01-31:
Thonny causes a logout as well, so my be my Mac or may be something in 10.14.6 - still to be investigated

JamesH65, 2022-11-14:
Is this still a problem? I did test on a Mac Mini when this was first released, and saw no issues

closed 2023-04-24 by JamesH65