raspberrypi/pico-project-generator #16: Setting Advanced Options Does not work

alphathegeek, 2021-01-31:
Setting any advanced option does not work:

Trying to set any name in Advanced Option gives the following error message:

NameError: name 'configs' is not defined
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "./pico_project.py", line 717, in OK
    DoEverything(self, p)
  File "./pico_project.py", line 1082, in DoEverything
    GenerateCMake('.', params)
  File "./pico_project.py", line 881, in GenerateCMake
    for c, v in configs.items():
NameError: name 'configs' is not defined

JamesH65, 2021-02-01:
Hmm, not sure how that one snuck through, looking at it now.

JamesH65, 2021-02-01:
OK, have pushed a fix to the master branch on the repo.

closed 2021-02-01 by JamesH65

alphathegeek, 2021-02-01:
Many thanks, tested ok! couldn't find the error myself on that one, thanks for fixing!