raspberrypi/pico-project-generator #20: Generate C++ option for Pico SDK

nyh-workshop, 2021-02-04:
Some of us are using C++ for this and need the team's support to add the Generate C++ option on that GUI (and also the command line)

JamesH65, 2021-02-10:
Should be added fixed with latest commit. Currently simply makes the project a .cpp file rather than just .c

closed 2021-02-10 by JamesH65

Trance-Paradox, 2021-06-14:
I am not able to build the project currently when using c++ file. Upon building the build config files are just added in the build folder and anything isn't compiled.

lurch, 2021-06-14:
Can you please provide more details? e.g. what OS are you using, what compiler are you using, what exact steps did you use, what command-line did you run, what error messages did you get, etc. etc.

Trance-Paradox, 2021-06-14:
> Can you please provide more details? e.g. what OS are you using, what compiler are you using, what exact steps did you use, what command-line did you run, what error messages did you get, etc. etc.

OS - WINDOWS 10
Command Line - MS Build Tools
IDE - VScode
Toolchain - GNU arm embedded

I opened the Pico Project generator and select the following settings
[image]

An then opened vscode from ms build tools command line. But building the project in release mode in vscode just give make files and make config in build folder. Any of the compiled files aren't created. That is there is no uf2 or hex file present in build folder.
[image]

lurch, 2021-06-14:
I see there's a `build.ninja` file in your screenshot - that's not right, and means you've missed one of the configuration steps. See the getting started guide https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

Trance-Paradox, 2021-06-15:
> I see there's a `build.ninja` file in your screenshot - that's not right, and means you've missed one of the configuration steps. See the getting started guide https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

Of I think I got it. I was configuring the environment again and again instead of building. But as you can see the bottom bar of my vscode does not contain any build button. How can I make it appear in the button bar?

lurch, 2021-06-15:
This is probably related to #32 and https://github.com/raspberrypi/pico-feedback/issues/134 
Unfortunately VS Code support is a low-priority task that we've not had enough spare time to deep-dive into.

It _should_ still work if you build from the command-line though?

Trance-Paradox, 2021-06-18:
> This is probably related to #32 and [raspberrypi/pico-feedback#134](https://github.com/raspberrypi/pico-feedback/issues/134)
> Unfortunately VS Code support is a low-priority task that we've not had enough spare time to deep-dive into.
> 
> It _should_ still work if you build from the command-line though?

Thanks for feedback