raspberrypi/pico-project-generator #21: [Request] Add support for picoprobe

kiwih, 2021-02-09:
A command line flag could be added so that the `launch.json` for VSCODE can use picoprobe instead of raspberrypi?

The output would need to become:
```
"configFiles": [
    "interface/picoprobe.cfg",
    "target/rp2040.cfg"
],
```

JamesH65, 2021-02-09:
Good idea, will do that when I next spend time on the generator, doing some other stuff at the moment.

JamesH65, 2021-02-10:
@liamfraser Is this all that is needed in the launch to make Picoprobe work with VSCode - presumably as an alternative to the current entry

```
      "configFiles": [
        "interface/raspberrypi-swd.cfg",
        "target/rp2040.cfg"
        ],

```

liamfraser, 2021-02-10:
Yes - that should be the only change you need

JamesH65, 2021-02-10:
I've added a PR that I think does what is needed. If @kiwih could give it a test that would be much appreciated.

https://github.com/raspberrypi/pico-project-generator/pull/22

kiwih, 2021-02-10:
Hi @JamesH65 I gave it a try but it wouldn't work for me until I made some other changes (This was true of the previous version as well.)

My working launch.json is:
```json
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [
    {
      "name": "Cortex Debug",
      "cwd": "${workspaceRoot}",
      "executable": "${command:cmake.launchTargetPath}",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "gdbPath" : "gdb-multiarch",
      "device": "Pico2040",
      "configFiles": [
        "interface/picoprobe.cfg",
        "target/rp2040.cfg"
        ],
      "svdFile": "/home/hammond/pico/pico-sdk/src/rp2040/hardware_regs/rp2040.svd",
      "runToMain": true,
    }
  ]
}
```

That is, I had to add gdbPath and change the executable argument. 
Is there a reason that your generated launch.json diverges from the one in the Getting Started guide? If I use the one from Getting Started it works (that was how I chose what the change these two arguments to).

kiwih, 2021-02-10:
Otherwise, your change works fine and the GUI is nice, so if the other things would be more appropriate for another issue that's fine as well

JamesH65, 2021-02-11:
Ah, thanks for that. Yes, the launch guide was updated after the last time I updated the generator. There was a change in the underlying Cortex Debug plugin. I'll fix the generator. We had a lot to do in the few days before release, this got missed, sorry!

JamesH65, 2021-02-11:
OK, have updated the generator to add gdbPath and fix the executable settings. If you could ensure it all works OK for you, I'll merge it.

kiwih, 2021-02-11:
Looking good - can I ask for one more QoL fix, can we change the svdFile line so that rather than generates

`"svdFile": "/home/hammond/pico/pico-sdk/src/rp2040/hardware_regs/rp2040.svd",`

it generates

`"svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",`

The reason I ask is that people will almost certainly end up committing these `launch.json`s to their version control / will share them, so if relative path names can be used (as they are in the Getting Started guide) they should probably also be used here.

Can the `device` also be changed to `RP2040` rather than `Pico2040`? That's also how it appears in the guide.

For your reference, here's the complete guide `launch.json`:
```json
{
  "version": "0.2.0",
  "configurations": [
      {
          "name": "Pico Debug",
          "cwd": "${workspaceRoot}",
          "executable": "${command:cmake.launchTargetPath}",
          "request": "launch",
          "type": "cortex-debug",
          "servertype": "openocd",
          // This may need to be arm-none-eabi-gdb depending on your system
          "gdbPath" : "gdb-multiarch",
          "device": "RP2040",
          "configFiles": [
              "interface/picoprobe.cfg",
              "target/rp2040.cfg"
          ],
          "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
          "runToMain": true,
          // Work around for stopping at main on restart
          "postRestartCommands": [
              "break main",
              "continue"
          ]
      }
  ]
}
```

I don't know how necessary the postRestartCommands are, you can probably skip those, they just add a breakpoint to main when you start debugging

JamesH65, 2021-02-11:
No problem, will do that now. I like the post command stuff - didn't know about those and I think a useful addition. Thanks for the help here! I just don't have time to dig around with all this stuff at the moment!

JamesH65, 2021-02-11:
I don't know much about this, but do the postrestartCommands suggested simply replicate the "runToMain" : "true"

JamesH65, 2021-02-11:
I've updated the PR, withou the post command stuff for the moment.

kiwih, 2021-02-11:
It is spot on perfect now. :+1:

kiwih, 2021-02-11:
I have tried with and without the postRestartCommands, my initial interpretation was incorrect - they affect usage of the "restart" button in the debugger. Without them, the restart button does not break on main (making it different to a normal launch, which does). 

If you include the 
```"postRestartCommands": [
              "break main",
              "continue"
          ]```
 it will make it so "starting" and "restarting" a session have the same behavior.

JamesH65, 2021-02-12:
I did wonder whether that might be the case.  I'll add it.

JamesH65, 2021-02-12:
I've pushed the change, if you are happy with it I will merge.

kiwih, 2021-02-12:
Works perfectly :+1:

kiwih, 2021-02-14:
I can see you've merged it now. Cheers!

closed 2021-02-14 by kiwih