raspberrypi/pico-project-generator #37: Can't get gui to work Ubuntu 20.04

thegreenalien, 2021-07-19:
after running the following code - 

sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential libssl-dev tk python3-tk

git clone https://github.com/raspberrypi/pico-project-generator.git

cd pico-project-generator

export PICO_SDK_PATH="/mnt/c/Users/john/pico/pico-sdk"

export DISPLAY=127.0.0.1:0

./pico_project.py --gui   

fails - 
Traceback (most recent call last):
  File "./pico_project.py", line 1245, in <module>
    RunGUI(sdkPath, args) # does not return, only exits
  File "./pico_project.py", line 242, in RunGUI
    root = tk.Tk()
  File "/usr/lib/python3.8/tkinter/__init__.py", line 2270, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display "127.0.0.1:0"

Any suggestions?

JamesH65, 2021-07-19:
Why have you set DISPLAY? IS that part of the instructions? (which instructions are you following?)

thegreenalien, 2021-07-19:
I'm following
https://github.com/garyexplains/examples/tree/master/Raspberry%20Pi%20Pico/C
but I've used the latest version of Ubuntu so that the version of cmake is suitable and doesn't need to be built separately.

closed 2021-07-19 by thegreenalien

lurch, 2021-07-19:
Just as an FYI, there's not strictly any need to use WSL, as https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf contains native instructions for Windows.