raspberrypi/pico-project-generator #45: pico_sdk_import.cmake not found

AlexChadwickP, 2021-11-29:
the generator can't seem to find pico_sdk_import.cmake for some reason although I've checked the path and it does exist. I'm running this from an WSL Ubuntu in Windows 11, here's a screenshot of the output and of the `ls` of the directory it's looking in:
[image]

joshuag1000, 2021-11-29:
i could be wrong but it looks like your pico path is set to ../pico-sdk try setting it to the exact path of where your pico sdk is.

AlexChadwickP, 2021-11-29:
> i could be wrong but it looks like your pico path is set to ../pico-sdk try setting it to the exact path of where your pico sdk is.

Yes, that was the problem, silly mistake from me, thanks!

closed 2021-11-29 by AlexChadwickP