raspberrypi/pico-project-generator #50: macOS TK Depreciation

tornupnegatives, 2022-02-28:
Running `pico_project.py` results in a completely black window being displayed, and the following warning is issued:

```
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.
```

I am running macOS 12.2.1 on a MacBook Pro (15-inch, 2019).

create-msugi, 2022-03-21:
You need to download Python3 from python.org and install it. Then everything will be fine.
I am using macOS 12.2.1 (Monterey) on a MacBook Pro (16-inch, 2019) and a Mac Pro (Late 2013).
It seems to be no problem.

lurch, 2022-03-21:
> You need to download Python3 from python.org and install it. Then everything will be fine.

That's interesting, thanks :+1: 

@JamesH65 Maybe it's worth mentioning that in this repo's README.md ?

aallan, 2022-03-21:
> You need to download Python3 from python.org and install it.

The recommended way to install Python is via [Homebrew](https://brew.sh/), see Section 9.1 in "[Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)".

tornupnegatives, 2022-03-21:
UPDATE: I was able to get it working by installing `tk-tcl` via Homebrew and then re-installing Python via pyenv. Finding a Python version that would build on macOS was a whole different can of worms, but at least the GUI is all good now :).

Closing since it is more of a Python/Tk problem than a RPi problem.

closed 2022-03-21 by tornupnegatives