raspberrypi/pico-project-generator #51: COM not working

StiglCZ, 2022-05-21:
Im using Windows 11 and ubuntu on WSL.
I was using the pico project generator and then test it out(the os i created) but the serial port not showing
When Im using other pico os(pogosm) its working just fine and serial port COM5 is existing.
My code was just hello world inside of while(true)
Compiled by going to build folder, cmake .. and then make

lurch, 2022-05-21:
Are you confusing the UART serial port with the USB-CDC serial port? See Chapter 4 in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf for more info.
And if you're still having problems, you should probably ask for help at https://forums.raspberrypi.com/viewforum.php?f=143

StiglCZ, 2022-05-21:
I did all things from the pdf but there is still not usb

StiglCZ, 2022-05-21:
Yes, I fixed this isssue by selecting all libraries in pico generator

closed 2022-05-21 by StiglCZ