raspberrypi/pico-project-generator #54: Option --project broken with commit last commit

Algafix, 2022-07-02:
Hi,

With the last commit, this line changed from:

`parser.add_argument("-p", "--project", action='append', help="Generate projects files for IDE. Options are: vscode")`

to:

`parser.add_argument("-p", "--project", action='append', default='vscode', help="Generate projects files for IDE. Options are: vscode")`

This has several implications.

1. The command `./pico_project.py -p vscode test` is broken. Output received:
```
$ ./pico_project.py -p vscode test
Traceback (most recent call last):
  File "/home/algafix/pico/pico-project-generator/./pico_project.py", line 1215, in <module>
    args = ParseCommandLine()
  File "/home/algafix/pico/pico-project-generator/./pico_project.py", line 831, in ParseCommandLine
    return parser.parse_args()
  File "/usr/lib/python3.9/argparse.py", line 1832, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 1865, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 2074, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.9/argparse.py", line 2014, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.9/argparse.py", line 1942, in take_action
    action(self, namespace, argument_values, option_string)
  File "/usr/lib/python3.9/argparse.py", line 1037, in __call__
    items.append(values)
AttributeError: 'str' object has no attribute 'append'
```
That's because the action is `action='append'`. Review the first answer from [here]( https://stackoverflow.com/questions/43660172/why-does-argparse-include-default-value-for-optional-argument-even-when-argument).

2. If no `-p` argument is specified, the script generates a bunch of `Unknown project type requested` error prints.
```
$ ./pico_project.py test          
PICO_SDK_PATH is /home/algafix/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is Debug
Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/algafix/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.9 (found version "3.9.7") found components: Interpreter 
TinyUSB available at /home/algafix/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Compiling TinyUSB with CFG_TUSB_DEBUG=1
cyw43-driver available at /home/algafix/pico/pico-sdk/lib/cyw43-driver
lwIP available at /home/algafix/pico/pico-sdk/lib/lwip
-- Configuring done
-- Generating done
-- Build files have been written to: /home/algafix/pico/pico-project-generator/test/build
Unknown project type requested
Unknown project type requested
Unknown project type requested
Unknown project type requested
Unknown project type requested
Unknown project type requested
```
That's because the code is expecting a list, and the `default` is providing an string. Thee code is then iterating on that string (line 1007).
```
    for p in projects :
        if p == 'vscode':
            foo
        else:
           print('Unknown project type requested')
```
3. The code is assuming that the project is vscode even if no `-p` argument is specified. This is related to how argparse handles the default option.

I would fix it myself and send a pull request. But because you were already trying to add some functionality with this last commit I prefer to let you make the decisions.

Hope it helps and thank you for this impressive utility.

Sorry about the typos, it's kinda late here,
Algafix.

lurch, 2022-07-02:
ping @aajshaw

aajshaw, 2022-07-03:
@Algafix thanks for spotting the problem, will be working on a fix. From your analysis it looks like just removing the `default='vscode'` will do the trick.

Algafix, 2022-07-03:
Indeed, it does :D 
Thanks for the project.

closed 2022-07-03 by Algafix