raspberrypi/pico-project-generator #59: Doesn't work with MinGW on Windows

BryanCrotaz, 2022-09-30:
I have the examples building successfully with MinGW and VSCode to avoid the multi-gigabyte Visual Studio Tools download.

However when I create my own project using pico-project-generator I get 

```
CMake Error at CMakeLists.txt:15 (project):
  Running 'nmake' '-?'
failed with:
  The system cannot find the file specified
```

BryanCrotaz, 2022-09-30:
I'm not enough of an expert in python to do this - needs a wake to find if `nmake` exists: 

```python
    if isWindows:
        if nmakeExists:
            cmakeCmd = 'cmake -DCMAKE_BUILD_TYPE=Debug -G "NMake Makefiles" ..'
            makeCmd = 'nmake '
        else:
            cmakeCmd = 'cmake -DCMAKE_BUILD_TYPE=Debug -G "MinGW Makefiles" ..'
            makeCmd = 'mingw32-make '
```

JamesH65, 2022-11-14:
Latest commit on master should support this. 203c601dd577160a4c518706b4d6ecc48d0ec367 I am unable to test this as I don't have a Windows system with MinGW, so if you have time to try it out that would be appreciated.

JamesH65, 2023-04-24:
Given no further reports of it failing after the fix, I'll close this issue. This area has also been updated to support Ninja.

closed 2023-04-24 by JamesH65