raspberrypi/pico-project-generator #67: Backslashes in compilerPath aren't escaped in VSCode JSON (Windows)

jpbloom, 2023-02-12:
The compilerPath populated in c_cpp_properties.json can cause a parsing error to be thrown in VSCode due to unescaped backslashes.

eg., 

`      "compilerPath": "C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\12.2 rel1\bin\arm-none-eabi-gcc.EXE",`

will fail, but

`      "compilerPath": "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\12.2 rel1\\bin\\arm-none-eabi-gcc.EXE",`

Is fine. Also, converting to forward slashes in the path works.

Might be particular to how certain system configs get the toolchain added to the system environment variables / PATH? Not sure, I'm a bit new to this toolchain.

Current workaround is to manually edit the JSON and then use CMake: Delete Cache and Reconfigure.

JamesH65, 2023-04-24:
I believe this is fixed on the `update_vscode_json_to_sdk1_5` branch, and I would welcome any testing.

ndabas, 2023-09-01:
That branch looks fine to me, but:
- I would use the `PICO_INSTALL_PATH` environment variable instead of hard-coding the path to the C/C++ compiler, so that any future updates to the installed SDK are automatically picked up and do not require a manual fix-up.
- I would also welcome discussion about keeping the VS Code config in a central location, mentioned previously here: https://github.com/raspberrypi/pico-project-generator/pull/87#issuecomment-1625114176