raspberrypi/pico-project-generator #68: Project Generator not compatible with SDK version 1.5 on Windows?

ullibak, 2023-04-06:
Hi
I installed the C/C++ SDK by running "pico-setup-windows-x64-standalone.exe" and then downloaded the Pico Project Generator. The Project Generator did not run because tkinter is missing (I posted this as a separate issue on github.com/raspberrypi/pico-setup-windows/issues/12).
I then installed a new version of Python manually and changed the python path in "pico-env.cmd" to point to that newly installed python.exe. With this trick, tkinter was present and I was able to run the Project Generator in GUI mode.

However, when trying to create a new project, the error message appears:

_CMake Error at CMakeLists.txt:21 (project):
  Running

   'nmake' '-?'

  failed with:

   Das System kann die angegebene Datei nicht finden  (=file not found)._

As far as I understand, the SDK installer for Windows sets up CMake to use ninja and not nmake (anymore). Is there a possibility to make the Project Generator compatible with the latest version of the SDK installer for Windows?

Thx.

JamesH65, 2023-04-06:
Hi, thanks for the report, I'll try and find the time to look in to this.

JamesH65, 2023-04-06:
@kilograham Hi Graham, do you know the runes for a ninja build on windows to replicate the current "cmake -DCMAKE_BUILD_TYPE=Debug -G "NMake Makefiles" .."

JamesH65, 2023-04-06:
I suspect it is as simple as :

cmake -DCMAKE_BUILD_TYPE=Debug -G Ninja ..

then 

`ninja` in the build folder?

kilograham, 2023-04-06:
yes (assuming the installer puts ninja in the path)

ullibak, 2023-04-06:
Modified pico_project.py starting from line 1341:

    if isWindows:
        # Had a special case report, when using MinGW, need to check if using nmake or mingw32-make.
        if shutil.which("mingw32-make"):
            # Assume MinGW environment
            cmakeCmd = 'cmake -DCMAKE_BUILD_TYPE=Debug -G "MinGW Makefiles" ..'
            makeCmd = 'mingw32-make '
        elif shutil.which("ninja"):
            # When installing SDK version 1.5.0 on windows with installer pico-setup-windows-x64-standalone.exe, ninja is used 
            cmakeCmd = 'cmake -DCMAKE_BUILD_TYPE=Debug -G Ninja ..'
            makeCmd = 'ninja '                 
        else:
            # Everything else assume nmake
            cmakeCmd = 'cmake -DCMAKE_BUILD_TYPE=Debug -G "NMake Makefiles" ..'
            makeCmd = 'nmake '

seems to work fine.

Many Thanks!

ullibak, 2023-04-08:
With the modifications described above, compiling was possible, but I ran into problems when trying to debug using VSCode. It turned out that the JSON files generated in the .vscode directory do not match the settings needed for the SDK installed using "pico-setup-windows-x64-standalone.exe". I modified the python script to add a function that generates the JSON files of the .vscode direcory in the examples folder. I am by far not an expert on these settings, so if someone with knowledge could have a look at it and maybe this code could be added in a future version of the pico project generator.

Here is the function:

```
# Generates the requested project files for Windows SDK 1.5.0 
def generateProjectFiles_Win(projectPath, projectName, sdkPath, projects, debugger):
```

    oldCWD = os.getcwd()

    os.chdir(projectPath)

    deb = debugger_config_list[debugger]

    for p in projects :
        if p == 'vscode':
            # launch.json
            v1 = (  '{\n'
                    '  "version": "0.2.0",\n'
                    '  "configurations": [\n'
                    '    {\n'
                    '      "name": "Pico Debug (Cortex-Debug)",\n'
                    '      "cwd": "${workspaceFolder}",\n'
                    '      "executable": "${command:cmake.launchTargetPath}",\n'
                    '      "request": "launch",\n'
                    '      "type": "cortex-debug",\n'
                    '      "servertype": "openocd",\n'
                    '      "gdbPath": "arm-none-eabi-gdb",\n'
                    '      "device": "RP2040",\n'
                    '      "configFiles": [\n'
                    '        "interface/cmsis-dap.cfg",\n'
                    '        "target/rp2040.cfg"\n'
                    '      ],\n'
                    '      "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",\n'
                    '      "runToEntryPoint": "main",\n'
                    '      "openOCDLaunchCommands": [\n'
                    '        "adapter speed 1000"\n'
                    '      ]\n'
                    '    },\n'
                    '    {\n'
                    '      "name": "Pico Debug (Cortex-Debug with external OpenOCD)",\n'
                    '      "cwd": "${workspaceFolder}",\n'
                    '      "executable": "${command:cmake.launchTargetPath}",\n'
                    '      "request": "launch",\n'
                    '      "type": "cortex-debug",\n'
                    '      "servertype": "external",\n'
                    '      "gdbTarget": "localhost:3333",\n'
                    '      "gdbPath": "arm-none-eabi-gdb",\n'
                    '      "device": "RP2040",\n'
                    '      "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",\n'
                    '      "runToEntryPoint": "main"\n'
                    '    },\n'
                    '    {\n'
                    '      "name": "Pico Debug (C++ Debugger)",\n'
                    '      "type": "cppdbg",\n'
                    '      "request": "launch",\n'
                    '      "cwd": "${workspaceFolder}",\n'
                    '      "program": "${command:cmake.launchTargetPath}",\n'
                    '      "MIMode": "gdb",\n'
                    '      "miDebuggerPath": "arm-none-eabi-gdb",\n'
                    '      "miDebuggerServerAddress": "localhost:3333",\n'
                    '      "debugServerPath": "openocd",\n'
                    '      "debugServerArgs": "-f interface/cmsis-dap.cfg -f target/rp2040.cfg -c \\"adapter speed 1000\\"",\n'
                    '      "serverStarted": "Listening on port .* for gdb connections",\n'
                    '      "filterStderr": true,\n'
                    '      "stopAtEntry": true,\n'
                    '      "hardwareBreakpoints": {\n'
                    '        "require": true,\n'
                    '        "limit": 4\n'
[... 128 lines cut ...]
            file.write(e1)
            file.close()
            
            file = open(VSCODE_CMAKEKITS_FILENAME, 'w')
            file.write(k1)
            file.close()
            
            file = open(VSCODE_TASKS_FILENAME, 'w')
            file.write(t1)
            file.close()
 
        else :
            print('Unknown project type requested')

    os.chdir(oldCWD)

And further down in the code, an if-clause to choose these files if on Windows:

```
if params['projects']:
        if isWindows:
            generateProjectFiles_Win(projectPath, params['projectName'], params['sdkPath'], params['projects'], params['debugger'])
        else:
            generateProjectFiles(projectPath, params['projectName'], params['sdkPath'], params['projects'], params['debugger'])
```

JamesH65, 2023-04-19:
Hi, I just pushed a possible fix for all this to

https://github.com/raspberrypi/pico-project-generator/tree/update_vscode_json_to_sdk1_5

Testing would be appreciated.

ullibak, 2023-04-22:
Tested. Test environment:
Windows Sandbox, freshly installed https://github.com/raspberrypi/pico-setup-windows 
Because tkinter is (for no obvious reasons) still missing in this packet, installed latest Python for Windows:
https://www.python.org/downloads/release/python-3113/
and edited "....\Pico SDK v1.5.0\pico-env.cmd" to change the path to the new python.exe:
```
rem call :AddToPath "%PICO_INSTALL_PATH%\python"
call :AddToPath "(something like: ....\AppData\Local\Programs\Python\Python311\)"
```

Started project generator in GUI mode, made a test project, opened VSCode, opened the directory of the test project and VSCode started working.
Error message appeared:
_Failed to parse "c:\pico\Test\Testproject1\.vscode\c_cpp_properties.json": Unexpected token p in JSON at position 4_

I can find no obvious errors in this JSON file, but I am not an expert on VSCode settings...

Klicked on CMake icon on the tool button and built the projekt. Everything ran smoothly with no error message. 
Project compiled and all the output files were generated.

JamesH65, 2023-04-24:
Looks to be a backslash/forward slash parsing issue on path names. Will take a look

JamesH65, 2023-04-24:
OK, this was actually a backslash escape issue, I think Windows paths need a double backslash in JSON files. So have added this and pushed to this branch for testing.