raspberrypi/pico-project-generator #71: PICO-SDK requirement  error message.

rewolff, 2023-05-04:
The at least version ... is required message when it is too old is saying 1.3.0 or later is required. 

I had 1.3.2-develop... Didn't work. 
I tried 1.3.1... Didn't work. 
I tried 1.3.0.... didn't work. 

I tried 1.4.0 and it did work. 

(I have a project that no longer compiles on 1.4.0 so I've had to go back a bit to find a working SDK for that project. So that's why I have older versions hanging around and don't like to "just go with 1.4.0".  Of course I'll "port" it to 1.4.0+ later on, but I don't have time to do that "now".)

lurch, 2023-05-04:
That appears to be related to this line of code https://github.com/raspberrypi/pico-project-generator/blob/master/pico_project.py#L1006

I can't see **anything** there saying 1.3.0 or later is required? :confused:

JamesH65, 2023-05-05:
Sounds like an older version of the generator is installed. It was recently updated to match the latest 1.5 SDK release and that line updated.

lurch, 2023-05-05:
Oh yes, looks like that was fixed in November last year https://github.com/raspberrypi/pico-project-generator/commit/19b3ac7e94fa9ce90f6451978ae2997c0068ba6b

rewolff, 2023-05-05:
I did a git pull on my pico-project-generator to be sure I was running the latest version. 

Ah! I was "on branch pico-w-support" has that been merged into master yet?

lurch, 2023-05-05:
> I was "on branch pico-w-support" has that been merged into master yet?

https://github.com/raspberrypi/pico-project-generator/branches says it has been merged :wink:

JamesH65, 2023-05-05:
Actually, 1.5 is the latest and greatest, so 1.3 is quite a long way behind. I'm surprised that it a problem upgrading, most is as before, there will only be a few minor changes in the API to make.

rewolff, 2023-05-05:
1.4 adds lwip to the SDK. 
I have a project that uses lwip, but wrote that before it was in the SDK. They clash.

rewolff, 2023-05-06:
FYI: (this is the wrong place to report this, but hopefully it lands in the right place) I did "git tag" and see a list of tags. I then assume that the bottom one (1.4) is the latest release so I do git checkout 1.4 to get what I thought was the "latest release". I'm guessing that someone forgot to add the tag for 1.5 ?

lurch, 2023-05-07:
> I did "git tag" and see a list of tags.

Which particular repo are you referring to?

rewolff, 2023-05-07:
Duh! I'm an idiot. I was talking about the pico-sdk repo. 
I didn't do the git pull on that. When I did: 
 * [new tag]         1.5.0               -> 1.5.0
Sorry for the noise.