raspberrypi/pico-project-generator #75: Support for Debug Probe

noutram, 2023-06-29:
I thought I should check with you before I try and do this myself :)

In terms of creating VSCode projects, I note that there are a couple options missing when generating launch.json

* Remote (whereby an openocd instance is already running and listening on port 3333)
* Debug Probe ( https://www.raspberrypi.com/products/debug-probe/ )

If would be really helpful if these were supported in this (rather useful) tool.

I've got working examples for both, but it's not as simple as a simple string substitution. My thinking would be to change the approach, and have some templates that are simply copied (as they do here https://github.com/raspberrypi/pico-examples/tree/master/ide/vscode ) into the generated project folder.

I'm considering writing the changes myself and generating a PR, but only if that sounds like an approach that might be acceptable?

Cheers,

Nick Outram, University of Plymouth, UK

noutram, 2023-06-29:
Note - for now - I've make a PR to support the debug probe https://github.com/raspberrypi/pico-project-generator/pull/76 as this is a simple addition to the launch.json file.

The above still stands however if _remote_ is to be added.

JamesH65, 2023-08-10:
PR is now merged. Leaving this open for the remote bit.