raspberrypi/pico-project-generator #88: Python not found

unitrix0, 2023-07-21:
I get this when I try to start it:

```
PS C:\Program Files\Raspberry Pi\Pico SDK v1.5.1> py C:\Users\mr\Documents\Pico-v1.5.1\pico-project-generator\pico_project.py --gui
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
```

even though I get a python shell when I execute `py` or `python` without parameters. I have no idea what I did wrong...

lurch, 2023-07-22:
ping @ndabas (as it looks like you might be using the Pico Windows Installer)

unitrix0, 2023-07-22:
Thats correct yes.

ndabas, 2023-07-22:
Okay that's a bit strange, but can you check a few things?
- Did you install the full version of Python separately from the Pico SDK installer? (The project generator does not work with the built-in version.)
- What do you get if you type `get-command py` and `get-command python` in the PowerShell prompt?
- Did you launch the PowerShell prompt you're using from the 'Pico - Developer PowerShell' shortcut?

unitrix0, 2023-07-22:
>     * Did you install the full version of Python separately from the Pico SDK installer? (The project generator does not work with the built-in version.)
I think so... I downloaded the Windows Installer from https://www.python.org/downloads/windows/

>     * What do you get if you type `get-command py` and `get-command python` in the PowerShell prompt?
```
PS C:\Program Files\Raspberry Pi\Pico SDK v1.5.1> Get-Command py | Format-Table -AutoSize

CommandType Name   Version        Source
----------- ----   -------        ------
Application py.exe 3.11.4150.1013 C:\Windows\py.exe

PS C:\Program Files\Raspberry Pi\Pico SDK v1.5.1> Get-Command python | Format-Table -AutoSize

CommandType Name       Version        Source
----------- ----       -------        ------
Application python.exe 3.9.13150.1013 C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\python\python.exe
```

>     * Did you launch the PowerShell prompt you're using from the 'Pico - Developer PowerShell' shortcut?
That was in the Pico Dev Shell

ndabas, 2023-07-22:
Thanks, looks like the Python version you installed isn't in the PATH -- but py.exe should have found it anyway (it doesn't use the PATH for that.)

Do you see Python 3.11 listed in the output of `py --list-paths`?

If not -- the solution for you might be to re-install the Python version that you downloaded, but make sure to check the option to add it to PATH.

unitrix0, 2023-07-22:
```
PS C:\Program Files\Raspberry Pi\Pico SDK v1.5.1> py --list-paths
 -V:3.11 *        C:\Program Files\Python311\python.exe
```
yes as it seems

ndabas, 2023-07-22:
Right, so the solution would be to re-install Python but make sure you add it to PATH this time, or just disable the app execution aliases for Python.

The underlying problem is that App Execution Aliases in Windows are messing around when `python` or `py` are called with arguments. From [Microsoft's FAQ on this topic](https://learn.microsoft.com/en-us/windows/python/faqs#why-does-running-python-exe-open-the-microsoft-store-):

> Running the shortcut executable with any command-line arguments will return an error code to indicate that Python was not installed. This is to prevent batch files and scripts from opening the Store app when it was probably not intended.
> 
> If you install Python using the installers from [python.org](https://www.python.org/downloads/windows/) and select the "add to PATH" option, the new python command will take priority over the shortcut. Note that other installers may add python at a lower priority than the built-in shortcut.
> 
> You can disable the shortcuts without installing Python by opening "Manage app execution aliases" from Start, finding the "App Installer" Python entries and switching them to "Off".

unitrix0, 2023-07-22:
so, the point is to re-install it?? because it is already in the PATH vriable
```
PS C:\Program Files\Raspberry Pi\Pico SDK v1.5.1> echo $env:Path.Split(';')
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\picotool
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk-tools
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\git\cmd
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\python
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\ninja
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\gcc-arm-none-eabi\bin
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\cmake\bin
C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\openocd
C:\Program Files\Python311\Scripts\
C:\Program Files\Python311\
C:\Windows\system32
C:\Windows
C:\Windows\System32\Wbem
C:\Windows\System32\WindowsPowerShell\v1.0\
C:\Windows\System32\OpenSSH\
C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\
C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\
C:\Program Files\Git\cmd
C:\Program Files (x86)\Microsoft SQL Server\150\Tools\Binn\
C:\Program Files\Microsoft SQL Server\150\Tools\Binn\
C:\Program Files\Microsoft SQL Server\150\DTS\Binn\
C:\Program Files (x86)\Microsoft SQL Server\150\DTS\Binn\
C:\Program Files\Azure Data Studio\bin
C:\Program Files\nodejs\
C:\Program Files\dotnet\
C:\Program Files\Microsoft SQL Server\130\Tools\Binn\
C:\Program Files\OpenSSL-Win64\bin

C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common
C:\WINDOWS\system32
C:\WINDOWS
C:\WINDOWS\System32\Wbem
C:\WINDOWS\System32\WindowsPowerShell\v1.0\
C:\WINDOWS\System32\OpenSSH\
C:\Program Files\Microsoft VS Code\bin
C:\Program Files (x86)\dotnet\
C:\Users\mr\AppData\Local\Microsoft\WindowsApps
C:\Program Files\JetBrains\WebStorm 2020.2.2\bin
C:\Program Files\Azure Data Studio\bin
C:\Users\mr\AppData\Roaming\npm
C:\Users\mr\.dotnet\tools
C:\Users\mr\AppData\Roaming\Python\Python311\Scripts
C:\Users\mr\AppData\Local\gitkraken\bi
```

unitrix0, 2023-07-24:
I tried starting it this way:
`& "C:\Program Files\Python311\python.exe" C:\users\mr\Documents\Pico-v1.5.1\pico-project-generator\pico_project.py --gui`
which kind of succeeded, the GUI came up, but when I created the project I got this error:

> PICO_SDK_PATH is C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk
> Defaulting PICO_PLATFORM to rp2040 since not specified.
> Defaulting PICO platform compiler to pico_arm_gcc since not specified.
> PICO compiler is pico_arm_gcc
> CMake Error at CMakeLists.txt:21 (project):
>   Running
> 
>    'nmake' '-?'
> 
>   failed with:
> 
>    Das System kann die angegebene Datei nicht finden
>  (The system cannot find the file specified)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "C:/_Repos/eWasserwaage/build/CMakeFiles/CMakeOutput.log".

The content of CMakeOutput.log:
> The target system is: PICO -  - cortex-m0plus
> The host system is: Windows - 10.0.19045 - AMD64

ndabas, 2023-07-24:
> so, the point is to re-install it?? because it is already in the PATH vriable

If it's already in the PATH, just try disabling the App Execution Alias as mentioned above.

It looks like the project was created fine but CMake used the wrong generator. You can try this from your project directory:
```pwsh
mkdir build # don't worry if this already exists
cd build
cmake -G Ninja .. --fresh
ninja
```

unitrix0, 2023-07-27:
well, it does build something, but when I copy it to the device, the display just stays black. And I can't neither build the examples... 

I guess I throw it in the corner or so..