raspberrypi/pico-project-generator #99: [windows] invalid compilerPath in c_cpp_properties.json

ychsiao168, 2023-08-27:
platform: win11

command: python pico_project.py --gui and "Create VSCode project" is checked.

line 10 in c_cpp_properties.json: 
```bash
"compilerPath": "C:\Program Files\Raspberry Pi\Pico SDK v1.5.1\gcc-arm-none-eabi\bin\arm-none-eabi-gcc.EXE",
```

currently I just modify '\\' to '\\\\' and everything is ok 

[image]

lurch, 2023-08-27:
Duplicate of #67 ? ping also @ndabas as it looks like you're using the Windows Pico Setup program.

ychsiao168, 2023-08-27:
> Duplicate of #67 ? ping also @ndabas as it looks like you're using the Windows Pico Setup program.

yes same as #67 and I'm using Windows Pico Setup program.

if I use 'update_vscode_json_to_sdk1_5' branch, compilerPath is ok.

closed 2023-08-27 by ychsiao168