raspberrypi/pico-project-generator #100: Enhancement: initialise git for the generated project

paul-lilley, 2023-09-06:
Thanks for an excellent tool.
Is there any appetite to extend what is generated to give a head-start for git within the generated project?
e.g.
- generate a basic .gitignore
- generate a stub for README.md (and maybe an empty LICENCE.txt)
- if we find a runnable git
    - git init in the generated project directory
    - an initial git commit of the generated files

If you think this would add value, I'm happy to put together a PR.

JamesH65, 2023-09-06:
That sounds useful! It would need to have a checkbox in the GUI and a command line option to match. I'd just have one checkbox that does all of the above.

paul-lilley, 2023-09-09:
Super, I'll put something together

paul-lilley, 2023-09-14:
@JamesH65 could you have a look at the PR when you get time? thanks